Benjamin-Dobell/Heimdall #497: timeout when calling libusb libusb_set_interface_alt_setting with some devices

gentooza, 2021-11-08:
Hi,

I've been suffering this error with my samsung mini plus (GT-S5570i) 
https://wiki.postmarketos.org/wiki/Samsung_Galaxy_Pop_Plus_(samsung-s5570i)

The only solution is to comment out the call to libusb_set_al_interface function, as said here https://github.com/Benjamin-Dobell/Heimdall/issues/340 by @joshgav.

As it seems not possible to detect phone's flaky bootloaders, because linux kernel at least seems to be samsung generic, a possible solution could be to set a command line parameter to bypass this function call. So if we execute heimdall and crash here, heimdall can hint to try out again with this bypass command.

The feedback and tests I've got from libusb hackers are here: 
https://github.com/libusb/libusb/issues/1024

I hope to be useful :-)

cheers!

Grimler91, 2021-11-22:
Hi @gentooza, heimdall seem to be working well without `libusb_set_interface_alt_setting` as far as I can tell on newer devices as well, so maybe we can remove it altogether. 

Could you at some point try flashing with [this heimdall](https://patchouli.sr.ht/builds.sr.ht/artifacts/~grimler/633164/61542720750d6e81/heimdall) (built from [this branch](https://git.sr.ht/~grimler/Heimdall/tree/alt_setting-error)). It will print the actual error code from libusb, and we can then, hopefully, tell heimdall to ignore that particular error code and carry on.

gentooza, 2021-11-22:
Hi @Grimler91

Ok, already cloned I'll test it and tell you

cheers

Grimler91, 2021-11-22:
@gentooza thanks! (Note that there are two branches in the repo, the master branch and the alt_setting-error branch. alt_setting-error contains [the commit](https://git.sr.ht/~grimler/Heimdall/commit/29aac0f5cb4ff32c8f69a3418630191fbcff9a68) with prints the libusb error code)

gentooza, 2021-11-23:
Hi @Grimler91, I paste what I get:

```
$ ./heimdall flash --RECOVERY /home/joa/dev/phones/samsung/galaxy\ mini\ s\ -\ gts5570i/boot.img 
Heimdall v1.4.2

Copyright (c) 2010-2017 Benjamin Dobell, Glass Echidna
https://www.glassechidna.com.au/

This software is provided free of charge. Copying and redistribution is
encouraged.

If you appreciate this software and you would like to support future
development please consider donating:
https://www.glassechidna.com.au/donate/

Initialising connection...
Detecting device...
Claiming interface...
Setting up interface...
libusb: error [op_set_interface] setintf failed error -1 errno 110
ERROR: Setting up interface failed with error code -99!
Releasing device interface...

$ git branch
* alt_setting-error
  master

$ git status
En la rama alt_setting-error
Tu rama está actualizada con 'origin/alt_setting-error'.

nada para hacer commit, el árbol de trabajo está limpio

$ git remote -v
origin	https://git.sr.ht/~grimler/Heimdall (fetch)
origin	https://git.sr.ht/~grimler/Heimdall (push)
```

Sorry, I've Spanish locale set in my GNU/Linux, the last commands are git orders simply to show I'm working in origin/alt_setting-error branch.

with libusb folks we were already trying to debug this and finally thought it's the flaky bootloader in the device.

ps: error -99 seems to be  LIBUSB_ERROR_OTHER  (https://libusb.sourceforge.io/api-1.0/group__libusb__misc.html)

Grimler91, 2021-12-14:
For future reference: the same issue was reported here: https://gitlab.com/BenjaminDobell/Heimdall/-/issues/517, and a similar fix (don't run libusb_set_interface_alt_setting if alt_setting is 0) was suggested.

gentooza, 2021-12-14:
the related merge request seems promising!
It has been approved? (sorry my head's going to explode between github, bitbucket, savannah, gitlab... GUIs)

cheers!

amo13, 2021-12-14:
I have created a [fork](https://github.com/amo13/Heimdall) and merged a few crucial pull requests. Feel free to create further pull requests there if need be.