Benjamin-Dobell/Heimdall #498: monterey support?

mrkprdo, 2021-11-24:
I just updated my mac, and was about to do stuff on my Android Tab when i downloaded and installed Heimdall, it is not supported yet with Monterey.

amo13, 2021-11-24:
This is an old and kind of abandoned project. But it appears to still be working fine with many devices. You need to call the binary directly from a terminal.

mrkprdo, 2021-11-24:
Yeah, i figured that this tool is already dated. Tried recompiling binaries but having issues with cmake, unable to find path for libusb although its there in Cellar lib.

nickelc, 2021-11-24:
A few months ago I wrote some tools, [frigg] and [wuotan], to play with my old devices.
wuotan is a flash tool inspired by Heimdall that works well to flash my GT-I9301I on Linux.

It doesn't have any quirks that seem to be required to work on Windows, but you may want to try it out.
It's written in Rust and allows to vendor libusb. I would be happy to get some feedback from a Mac user.  

[frigg]: https://github.com/nickelc/frigg
[wuotan]: https://github.com/nickelc/wuotan

mrkprdo, 2021-11-24:
I used windows instead. I think if someone is having issues with this tool, use the mentioned tools in the above comment as an alternative.

closed 2021-11-24 by mrkprdo

amo13, 2021-11-25:
I'm kind of a bit late, but I made {a fork](https://github.com/amo13/Heimdall) and merged a few crucial pull requests. No need for compiling, you can grab the latest binaries compiled with github actions from the releases there.