Benjamin-Dobell/Heimdall #502: Question: Flash from compressed file

realivanjx, 2021-12-03:
Is it required to know/send total uncompressed file size in the protocol? I have flash files with lz4 compression but i want to be able to flash it without needing to extract it fully. Thanks.

Grimler91, 2021-12-14:
You can't send the compressed file to the device, but it could be automatically extracted on-the-fly by heimdall

realivanjx, 2021-12-14:
@Grimler91 but is there a way to not send the total flash size since i need to decompress everything before sending it to the device? .tar.md5 as far as i know uses lz4 encryption and does not store the uncompressed size.

Grimler91, 2021-12-14:
Apparently the size is there in the header:

```
$ lz4 --list system.img.lz4
    Frames           Type Block  Compressed  Uncompressed     Ratio   Filename
         1       LZ4Frame   B6I       2.43G         3.77G     64.39%  system.img.lz4
```
It is generally not included in the header, as you said:
```
$ lz4 foo foo.lz4
Compressed 69737165 bytes into 69727297 bytes ==> 99.99%                       
$ lz4 --list foo.lz4 
    Frames           Type Block  Compressed  Uncompressed     Ratio   Filename
         1       LZ4Frame   B7I      66.50M             -         -   foo.lz4
```

lz4 needs to be invoked with `--content-size` for the information to be included, it seems that `-B6` is also needed to get an archive identical to Samsung's (based on https://github.com/lz4/lz4/issues/802)

realivanjx, 2021-12-14:
thank you very much for the pointer @Grimler91. i will try to just read it from the lz4 header and if it does not exist i will just throw an error.

closed 2021-12-14 by realivanjx