Benjamin-Dobell/Heimdall #503: [SM-G9700] S10e - can't flash recovery

lannic, 2021-12-20:
Archlinux - Heimdall v1.4.2 - libusb-1.0.24-2

I can't upload the recovery lineageos image to the phone.
I got these errors in the end of the process : 

ERROR: Failed to unpack received packet.
ERROR: Failed to receive session end confirmation!

I attach PIT file and recovery trace
[recovery.txt](https://github.com/Benjamin-Dobell/Heimdall/files/7746473/recovery.txt)
[pit.txt](https://github.com/Benjamin-Dobell/Heimdall/files/7746480/pit.txt)

amo13, 2021-12-20:
1. What file did you try to send?
2. What command did you execute?
3. Have you tried a heimdall binary compiled with [a few crucial fixes](https://github.com/amo13/Heimdall/releases)?

lannic, 2021-12-20:
What file did you try to send?
this one : [https://mirrorbits.lineageos.org/recovery/beyond0lte/20211216/lineage-18.1-20211216-recovery-beyond0lte.img](url)

What command did you execute?
heimdall-linux flash --RECOVERY lineage-18.1-20211216-recovery-beyond0lte.img --no-reboot

Have you tried a heimdall binary compiled with a few crucial fixes?
Yes same problem

amo13, 2021-12-20:
OK, then I'd also like to know why it is not working

amo13, 2021-12-20:
Have you found the similar issue on this repo with the following hint?

I had the same errors, the only known fix for me (as of the USB3 port problem) was to reboot the device into Download Mode after each time I did another heimdall command. I know, it's totally annoying, but that way I could flash the recovery partition with TWRP and use sideload later.

lannic, 2021-12-20:
For upload the recovery image, i shutdown totally the phone and boot in the Download mode.
Then, in this mode, i execute the command for flash the recovery partition

Grimler91, 2021-12-20:
It says

> RECOVERY upload successful

And then seem to just fail to exit. Are you sure it hasn't been flashed?

lannic, 2021-12-20:
Yes, it hasn't been flashed because i don't have the lineageos picture but the samsung menu

Grimler91, 2021-12-20:
What OS are you on, and have you tried without the ` --no-reboot` flag?

lannic, 2021-12-20:
I'm on Archlinux with a fresh update
Same problem without the --no-reboot flag

Grimler91, 2021-12-20:
Where does your `heimdall-linux` come from? Neither the [heimdall package](https://archlinux.org/packages/community/x86_64/heimdall/) nor the [heimdall-git aur package](https://aur.archlinux.org/packages/heimdall-git/) installs heimdall as `heimdall-linux`.

Could you try [this heimdall](https://patchouli.sr.ht/builds.sr.ht/artifacts/~grimler/653348/d981d2e3eae5626f/heimdall) as well? If that doesn't work either then we probably need captured usb-traffic logs from wireshark of when odin flashes stock android (or at least recovery), to see if the protocol has changed in some way for this device.

(I haven't had any similar problems on galaxy s8 on archlinux, don't have any newer devices to test on)

Grimler91, 2021-12-20:
Device also needs to be OEM unlocked for it to accept custom recovery/roms

lannic, 2021-12-21:
> Device also needs to be OEM unlocked for it to accept custom recovery/roms
For sure

> Could you try this heimdall as well? 
The end of return is a little bit different : 
Uploading RECOVERY
100%
RECOVERY upload successful

Ending session...
ERROR: Failed to receive session end confirmation!
Releasing device interface...

> If that doesn't work either then we probably need captured usb-traffic logs from wireshark
Here the link for the usb capture [https://numeriquement.fr/nextcloud/index.php/s/J88aQwZJ2Bbf9Lr](url)

Grimler91, 2021-12-21:
@lannic thanks, that's the log of failed recovery flash with heimdall, right? 
We probably need logs of successful flash with [Odin](https://www.odinflash.com/) to be able to determine what heimdall does differently compared to odin for this device.
To create an Odin-compatible tar of recovery.img that you can flash (as an AP file in odin) you can do:
```
tar cf recovery.img.tar recovery.img
cat recovery.img.tar <(md5sum recovery.img.tar) > recovery.img.tar.md5
```

lannic, 2021-12-21:
> that's the log of failed recovery flash with heimdall, right?

Yes

Here a link of the 7z recovery.img.tar.md5
https://numeriquement.fr/nextcloud/index.php/f/122930

Grimler91, 2021-12-21:
Do you have the possibility to flash it with odin on windows and capture logs?

lannic, 2021-12-21:
No. I don't have any windows

Grimler91, 2021-12-21:
Alright, then we'll have to wait and see if anyone else has the same problem and the possibility to capture logs

lannic, 2022-10-18:
You can try maybe it's OK now ?

panomitrius, 2022-10-18:
Wouldn't that entail a risk of bricking the phone?

lannic, 2022-10-18:
No. No problem.
Because when you restart the phone, you cannot install the linageos rom.
See this part of installation : [https://wiki.lineageos.org/devices/beyond0lte/install#installing-lineageos-from-recovery](url)

panomitrius, 2022-11-10:
Ok, great, thanks!

LAfricain, 2023-12-14:
I have the same problem with an Samsung A40.

ulidtko, 2026-08-17:
Hit the same issue with `CHN`-market SM-T580. It was from **OEM-locked bootloader**.

To unlock, there're TWO steps:
1. In developer options — just like every guide says — tick the "OEM unlock enabled" checkbox. What most guides fail to mention: that checkbox *allows* the unlock to be done; but does no unlock by itself.
2. Install [`CROM_Service_1.0.7.apk`](https://archive.org/details/crom-service-1.0.7) and run it — this is what *actually unlocks* the bootloader on Samsung devices.

After that, I got a clean flash with Heimdall, `RECOVERY upload successful` — `Ending session` — `Releasing device interface` with no errors. Whereas before (with bootloader locked), it was saying `ERROR: Failed to receive session end confirmation!` — and the bootloader on device was showing a page saying *You need to unlock your device to install a custom operating system* (in Chinese characters).

Another pitfall worth reminding here: the stock ROMs will by default *revert* the recovery image on boot into normal OS. This is the reason one has to supply `--no-reboot` in `heimdall` command line, to preempt that. In my case — flashing TWRP recovery — this allowed to reboot into TWRP (which broke off the default reverting logic), and allowed it to stay put.

Hope that helps.