Benjamin-Dobell/Heimdall #504: [SM-F926B] ERROR: Failed to send handshake!

Azkali, 2021-12-24:
# Issue description

Heimdall version: Latest master build from source

During `./heimdall download-pit` handshake is failing. Seems related to changes in download mode on recent model, as hinted by another issue here.

Log :
```
Heimdall v1.4.2

Copyright (c) 2010-2017 Benjamin Dobell, Glass Echidna
http://www.glassechidna.com.au/

This software is provided free of charge. Copying and redistribution is
encouraged.

If you appreciate this software and you would like to support future
development please consider donating:
http://www.glassechidna.com.au/donate/

Initialising connection...
Detecting device...
[timestamp] [threadID] facility level [function call] <message>
--------------------------------------------------------------------------------
[ 0.016216] [0003d102] libusb: debug [libusb_get_device_list]  
[ 0.016249] [0003d102] libusb: debug [libusb_get_device_descriptor]  
[ 0.016262] [0003d102] libusb: debug [libusb_get_device_descriptor]  
[ 0.016276] [0003d102] libusb: debug [libusb_open] open 1.26
[ 0.016318] [0003d102] libusb: debug [usbi_add_event_source] add fd 8 events 4
[ 0.016331] [0003d102] libusb: debug [libusb_get_device_descriptor]  
[ 0.016344] [0003d102] libusb: debug [libusb_get_config_descriptor] index 0
Claiming interface...
[ 0.016369] [0003d102] libusb: debug [libusb_claim_interface] interface 1
Setting up interface...
[ 0.016427] [0003d102] libusb: debug [libusb_set_interface_alt_setting] interface 1 altsetting 0

Initialising protocol...
[ 0.017238] [0003d102] libusb: debug [libusb_alloc_transfer] transfer 0x493840
[ 0.017256] [0003d102] libusb: debug [libusb_submit_transfer] transfer 0x493840
[ 0.017266] [0003d102] libusb: debug [add_to_flying_list] arm timer for timeout in 1000ms (first in line)
[ 0.017281] [0003d102] libusb: debug [submit_bulk_transfer] need 1 urbs for new transfer with length 4
[ 0.017309] [0003d102] libusb: debug [libusb_handle_events_timeout_completed] doing our own event handling
[ 0.017320] [0003d102] libusb: debug [handle_events] event sources modified, reallocating event data
[ 0.017330] [0003d102] libusb: debug [usbi_wait_for_events] poll() 3 fds with timeout in 60000ms
[ 1.017435] [0003d102] libusb: debug [usbi_wait_for_events] poll() returned 1
[ 1.017749] [0003d102] libusb: debug [libusb_cancel_transfer] transfer 0x493840
[ 1.018411] [0003d102] libusb: debug [arm_timer_for_next_timeout] no timeouts, disarming timer
[ 1.018670] [0003d102] libusb: debug [libusb_handle_events_timeout_completed] doing our own event handling
[ 1.018758] [0003d102] libusb: debug [usbi_wait_for_events] poll() 3 fds with timeout in 60000ms
[ 1.019017] [0003d102] libusb: debug [usbi_wait_for_events] poll() returned 1
[ 1.019302] [0003d102] libusb: debug [reap_for_handle] urb type=3 status=-2 transferred=0
[ 1.019971] [0003d102] libusb: debug [handle_bulk_completion] handling completion status -2 of bulk urb 1/1
[ 1.020094] [0003d102] libusb: debug [handle_bulk_completion] abnormal reap: urb status -2
[ 1.020183] [0003d102] libusb: debug [handle_bulk_completion] abnormal reap: last URB handled, reporting
[ 1.020282] [0003d102] libusb: debug [usbi_handle_transfer_cancellation] detected timeout cancellation
[ 1.020378] [0003d102] libusb: debug [arm_timer_for_next_timeout] no timeouts, disarming timer
[ 1.020529] [0003d102] libusb: debug [usbi_handle_transfer_completion] transfer 0x493840 has callback 0x7fa3bc49e0
[ 1.020644] [0003d102] libusb: debug [sync_transfer_cb] actual_length=0
[ 1.020768] [0003d102] libusb: debug [libusb_free_transfer] transfer 0x493840
[ 1.271112] [0003d102] libusb: debug [libusb_alloc_transfer] transfer 0x474f10
[ 1.271303] [0003d102] libusb: debug [libusb_submit_transfer] transfer 0x474f10
[ 1.271394] [0003d102] libusb: debug [add_to_flying_list] arm timer for timeout in 1000ms (first in line)
[ 1.271670] [0003d102] libusb: debug [submit_bulk_transfer] need 1 urbs for new transfer with length 4
[ 1.271959] [0003d102] libusb: debug [libusb_handle_events_timeout_completed] doing our own event handling
[ 1.272068] [0003d102] libusb: debug [usbi_wait_for_events] poll() 3 fds with timeout in 60000ms
[ 2.272444] [0003d102] libusb: debug [usbi_wait_for_events] poll() returned 1
[ 2.272516] [0003d102] libusb: debug [libusb_cancel_transfer] transfer 0x474f10
[ 2.272895] [0003d102] libusb: debug [arm_timer_for_next_timeout] no timeouts, disarming timer
[ 2.272926] [0003d102] libusb: debug [libusb_handle_events_timeout_completed] doing our own event handling
[ 2.272939] [0003d102] libusb: debug [usbi_wait_for_events] poll() 3 fds with timeout in 60000ms
[... 114 lines cut ...]
[10.793937] [0003d102] libusb: debug [libusb_free_transfer] transfer 0x48d0b0
ERROR: Protocol initialisation failed!

Releasing device interface...
[10.794238] [0003d102] libusb: debug [libusb_release_interface] interface 1

[10.794615] [0003d102] libusb: debug [libusb_close]  
[10.794715] [0003d102] libusb: debug [usbi_remove_event_source] remove fd 8
[10.794856] [0003d102] libusb: debug [libusb_exit]  
[10.794901] [0003d102] libusb: debug [libusb_exit] destroying default context
[10.794944] [0003d102] libusb: debug [libusb_handle_events_timeout_completed] doing our own event handling
[10.794981] [0003d102] libusb: debug [handle_events] event sources modified, reallocating event data
[10.795027] [0003d102] libusb: debug [usbi_wait_for_events] poll() 2 fds with timeout in 0ms
[10.795075] [0003d102] libusb: debug [usbi_wait_for_events] poll() returned 0
[10.795119] [0003d102] libusb: debug [libusb_unref_device] destroy device 2.1
[10.795164] [0003d102] libusb: debug [libusb_unref_device] destroy device 1.26
[10.795209] [0003d102] libusb: debug [libusb_unref_device] destroy device 1.13
[10.795255] [0003d102] libusb: debug [libusb_unref_device] destroy device 1.12
[10.795299] [0003d102] libusb: debug [libusb_unref_device] destroy device 1.11
[10.795342] [0003d102] libusb: debug [libusb_unref_device] destroy device 1.10
[10.795387] [0003d102] libusb: debug [libusb_unref_device] destroy device 1.1
[10.795435] [0003d102] libusb: debug [usbi_remove_event_source] remove fd 7
[10.795572] [0003d102] libusb: debug [usbi_remove_event_source] remove fd 6
[10.795761] [0003d104] libusb: debug [linux_udev_event_thread_main] udev event thread exiting
```

Grimler91, 2021-12-26:
Are you on Ubuntu? It fails on the first handshake package, so it is unlikely that it is due to protocol changes (unless Samsung has re-done the protocol from scratch for this device). Seems more likely that the connection needs to be reset before handshake, as in [this PR](https://github.com/Benjamin-Dobell/Heimdall/pull/478). You can download heimdall with this (and other) fixes here: [ubuntu 20.04](https://patchouli.sr.ht/builds.sr.ht/artifacts/~grimler/650736/bf10d326fa210f7b/heimdall), [archlinux](https://patchouli.sr.ht/builds.sr.ht/artifacts/~grimler/650735/07ba1155b5754e4a/heimdall) and [alpine linux](https://patchouli.sr.ht/builds.sr.ht/artifacts/~grimler/650734/8a998383f5f18428/heimdall).

Azkali, 2022-01-04:
Hi, I am using fedora 35 on a aarch64 SoC. But I was able to use Heimdall with a SGS8 with no problem previously.

The issue also happen as root and I do have the udev rules set and triggered properly.

I am gonna check the PR and give my feedback after testing.

Grimler91, 2022-01-04:
Apparently [odin 3.14.4](https://forum.xda-developers.com/t/bul6-oneui-4-0-official-stock-fw-jan-22-z-fold3-international-sm-f926b.4383005/) works for flashing this device, so seems unlikely that protocol has been fundamentally changed.

However, looking at the PIT is seems that it uses a 6th version of the Odin protocol (first time I see that, I only have devices using 4 or older):
```
$ heimdall print-pit --file Q2Q_EUR_OPENX.pit
Heimdall v2.0.0

Copyright (c) 2010-2017 Benjamin Dobell, Glass Echidna
https://www.glassechidna.com.au/

This software is provided free of charge. Copying and redistribution is
encouraged.

If you appreciate this software and you would like to support future
development please consider donating:
https://www.glassechidna.com.au/donate/

--- PIT Header ---
Entry Count: 96
Unknown string: COM_TAR2
CPU/bootloader tag: SM8350
Version(?): 0x0006

[...]
```
(printed with heimdall from [this branch](https://git.sr.ht/~grimler/Heimdall)) so maybe something is different, but then again: seems unlikely they would change something as fundamental as the handshake. 

Would be interesting to see usb traffic logs (captured with for example [wireshark](https://wiki.wireshark.org/CaptureSetup/USB)) of when stock android is flashed onto the device with odin. Such logs would be the only way to see if the transfer protocol is different.

Grimler91, 2022-01-04:
You can also try increasing the timeout (third arg to SendBulkTransfer) of the [handshake package](https://github.com/Benjamin-Dobell/Heimdall/blob/master/heimdall/source/BridgeManager.cpp#L306), and probably [the response](https://github.com/Benjamin-Dobell/Heimdall/blob/master/heimdall/source/BridgeManager.cpp#L316), to something higher like 6000.

Really seems more likely that issue is that libusb_reset is needed, or some issue with libusb on feodora/arm64, or just that the usb cable is bad, rather than that heimdall does something wrong (since heimdall doesn't even do anything, first packet fails)

alvsz, 2022-02-21:
> You can download heimdall with this (and other) fixes here: [ubuntu 20.04](https://patchouli.sr.ht/builds.sr.ht/artifacts/~grimler/650736/bf10d326fa210f7b/heimdall), [...]

omg thank you i was having the same issue this took me so long to find

TheAirBlow, 2022-05-27:
> Apparently [odin 3.14.4](https://forum.xda-developers.com/t/bul6-oneui-4-0-official-stock-fw-jan-22-z-fold3-international-sm-f926b.4383005/) works for flashing this device, so seems unlikely that [...]

It is not version of some sort. Odin, nor Thor (it's successor) use data after entries count, Everything after is referenced as Project Name.

Azkali, 2022-09-29:
Solved using the PR mentioned in the previous comment

pcolamar, 2022-10-25:
I am in the same situation. Linux Mint 21 (aka Ubuntu 22 LTS) and the heimdall in the standard repo (1.4.1) does not work.
"ERROR: Failed to send handshake!ERROR: Protocol initialisation failed! "

With the 2.0.0 of the link below , it works.

I guess I should report this to Mint folks ( or Ubuntu 22).

TheAirBlow, 2022-10-28:
> I am in the same situation. Linux Mint 21 (aka Ubuntu 22 LTS) and the heimdall in the standard repo (1.4.1) does not work. "ERROR: Failed to send handshake!ERROR: Protocol initialisation failed! " [...]

No, it is an unofficial fork. Maybe there is a chance tgey will use a fork, but who knows.

julianfoad, 2024-05-25:
Duplicate of #413? I just added some help there.

T-vK, 2024-09-01:
On Manjaro, I managed to get it to work by uninstalling `heimdall` from `Official Repositories (extra)` and then installing `heimdall-git` from `AUR`.

VictorQueiroz, 2024-11-08:
In my case, I noticed that when in download mode, you can only flash once. If there is a pending operation, it will remain pending until you restart the device. The `--resume` argument does not work on my **SM-A546E/DS**.

The easiest way I found to deal with this while desperately trying to root the phone as soon as possible is by doing the following:

1. have the phone connected via USB
2. Run the `watch—n .5 adb reboot download` command continuously on the terminal until the phone goes into download mode again.
3. Press the side key and volume down until the phone restarts; for some reason, the `adb` command will work when the phone is booting.
4. Stop the `watch` command
5. Try to flash correctly this time

@T-vK `heimdall-git` does seem to be more stable when compared to the `heimdall` package.

rklasen, 2025-03-05:
Can confirm `heimdall-git` to be working, whereas `heimdall` didn't.