Benjamin-Dobell/Heimdall #509: libusb: error [get_usbfs_fd] libusb couldn't open USB device /dev/bus/005/020, errno=5

dfarmilant, 2022-02-17:
I'm attempting to root an SM-P600 Galaxy 10.1 2014 tablet. I'm able to detect a device is via heimdall-frontend -> utilities -> detect.

If I attempt to download or print the PIT file in the CLI, I get the following output. I get a similar output to the one below, less the line with the "###".

```
[def@tower bin]$ sudo ./heimdall print-pit --no-reboot
Heimdall v1.4.2

Copyright (c) 2010-2017 Benjamin Dobell, Glass Echidna
http://www.glassechidna.com.au/

This software is provided free of charge. Copying and redistribution is
encouraged.

If you appreciate this software and you would like to support future
development please consider donating:
http://www.glassechidna.com.au/donate/

Initialising connection...
Detecting device...
libusb: error [get_usbfs_fd] libusb couldn't open USB device /dev/bus/usb/005/020, errno=5   ###
ERROR: Failed to access device. libusb error: -1
```
I've searched this and the libusb-git repos and haven't been found a solution. Any tips?

Grimler91, 2022-02-17:
Have you tried without sudo, or with another usb cable?

dfarmilant, 2022-02-17:
Welp, it may not matter if that's the fix now. I may have bricked it. I'll let you know if that works.

dfarmilant, 2022-02-18:
@Grimler91 I've changed the USB cable and ran without sudo privileges. When I start the tablet in download mode, I see:

```
ODIN MODE
PRODUCT NAME: SM-P600
CURRENT BINARY: Samsung Official
SYSTEM STATUS: Official
Secure Download: Enabled
KNOX WARRANTY VOID: 1
RP SWREV: A3
```
Is there a specific USB cable I should use?

dfarmilant, 2022-02-19:
I was having trouble with a USB drive connecting and realized I hadn't restarted my computer after a kernel update (bad form, I know, but I'll always do it now!). That didn't solve it.

However, I found [this](https://github.com/aguegu/stm8demos#troubleshooting) when searching for the `[get_usbfs_fd]` error. It seems relevant. In the event that it is, I suppose I would need the `ATTR{idProduct}` code and add that to the rules file.

@Grimler91 thoughts?

Grimler91, 2022-02-19:
What kind of device, and OS, are you running heimdall on? 

> However, I found [this](https://github.com/aguegu/stm8demos#troubleshooting) when searching for the `[get_usbfs_fd]` error. It seems relevant. In the event that it is, I suppose I would need the `ATTR{idProduct}` code and add that to the rules file.

Udev rules could help, but I think `sudo` works around such issues (and that didn't help for you).

What information about the device does `lsusb -v` show?

dfarmilant, 2022-02-20:
This is a desktop computer running Manjaro 5.16.7-1.

```
Bus 006 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Device Descriptor:
  bLength                18
  bDescriptorType         1
  bcdUSB               3.10
  bDeviceClass            9 Hub
  bDeviceSubClass         0 
  bDeviceProtocol         3 
  bMaxPacketSize0         9
  idVendor           0x1d6b Linux Foundation
  idProduct          0x0003 3.0 root hub
  bcdDevice            5.16
  iManufacturer           3 Linux 5.16.7-1-MANJARO xhci-hcd
  iProduct                2 xHCI Host Controller
  iSerial                 1 0000:0e:00.3
  bNumConfigurations      1
  Configuration Descriptor:
    bLength                 9
    bDescriptorType         2
    wTotalLength       0x001f
    bNumInterfaces          1
    bConfigurationValue     1
    iConfiguration          0 
    bmAttributes         0xe0
      Self Powered
      Remote Wakeup
    MaxPower                0mA
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        0
      bAlternateSetting       0
      bNumEndpoints           1
      bInterfaceClass         9 Hub
      bInterfaceSubClass      0 
      bInterfaceProtocol      0 Full speed (or root) hub
      iInterface              0 
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x81  EP 1 IN
        bmAttributes            3
          Transfer Type            Interrupt
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0004  1x 4 bytes
        bInterval              12
        bMaxBurst               0
Hub Descriptor:
  bLength              12
  bDescriptorType      42
  nNbrPorts             4
  wHubCharacteristic 0x000a
    No power switching (usb 1.0)
    Per-port overcurrent protection
  bPwrOn2PwrGood       50 * 2 milli seconds
  bHubContrCurrent      0 milli Ampere
  bHubDecLat          0.0 micro seconds
  wHubDelay             0 nano seconds
  DeviceRemovable    0x00
 Hub Port Status:
   Port 1: 0000.02a0 lowspeed L1
   Port 2: 0000.02a0 lowspeed L1
   Port 3: 0000.02a0 lowspeed L1
   Port 4: 0000.02a0 lowspeed L1
Binary Object Store Descriptor:
  bLength                 5
[... 1852 lines cut ...]
          Usage Type               Data
        wMaxPacketSize     0x0004  1x 4 bytes
        bInterval              12
Hub Descriptor:
  bLength               9
  bDescriptorType      41
  nNbrPorts             6
  wHubCharacteristic 0x000a
    No power switching (usb 1.0)
    Per-port overcurrent protection
    TT think time 8 FS bits
  bPwrOn2PwrGood       10 * 2 milli seconds
  bHubContrCurrent      0 milli Ampere
  DeviceRemovable    0x10
  PortPwrCtrlMask    0xff
 Hub Port Status:
   Port 1: 0000.0100 power
   Port 2: 0000.0100 power
   Port 3: 0000.0100 power
   Port 4: 0000.0103 power enable connect
   Port 5: 0000.0100 power
   Port 6: 0000.0100 power
Device Status:     0x0001
  Self Powered
```

dfarmilant, 2022-02-20:
This is a custom desktop running Manjaro 15.16.7-1.

Here's `lsusb -v` for the tablet:

```
Bus 001 Device 011: ID 04e8:685d Samsung Electronics Co., Ltd GT-I9100 Phone [Galaxy S II] (Download mode)
Device Descriptor:
  bLength                18
  bDescriptorType         1
  bcdUSB               2.00
  bDeviceClass            2 Communications
  bDeviceSubClass         2 Abstract (modem)
  bDeviceProtocol         0 
  bMaxPacketSize0        64
  idVendor           0x04e8 Samsung Electronics Co., Ltd
  idProduct          0x685d GT-I9100 Phone [Galaxy S II] (Download mode)
  bcdDevice            2.1b
  iManufacturer           1 SAMSUNG
  iProduct                2 Gadget Serial
  iSerial                 0 
  bNumConfigurations      1
  Configuration Descriptor:
    bLength                 9
    bDescriptorType         2
    wTotalLength       0x0043
    bNumInterfaces          2
    bConfigurationValue     1
    iConfiguration          0 
    bmAttributes         0xc0
      Self Powered
    MaxPower               50mA
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        0
      bAlternateSetting       0
      bNumEndpoints           1
      bInterfaceClass         2 Communications
      bInterfaceSubClass      2 Abstract (modem)
      bInterfaceProtocol      1 AT-commands (v.25ter)
      iInterface              0 
      CDC Header:
        bcdCDC               1.10
      CDC Call Management:
        bmCapabilities       0x00
        bDataInterface          1
      CDC ACM:
        bmCapabilities       0x00
      CDC Union:
        bMasterInterface        0
        bSlaveInterface         1 
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x83  EP 3 IN
        bmAttributes            3
          Transfer Type            Interrupt
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0010  1x 16 bytes
        bInterval               9
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        1
      bAlternateSetting       0
      bNumEndpoints           2
      bInterfaceClass        10 CDC Data
      bInterfaceSubClass      0 
      bInterfaceProtocol      0 
      iInterface              1 
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x81  EP 1 IN
        bmAttributes            2
          Transfer Type            Bulk
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0200  1x 512 bytes
        bInterval               0
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x02  EP 2 OUT
        bmAttributes            2
          Transfer Type            Bulk
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0200  1x 512 bytes
        bInterval               0
```

dfarmilant, 2022-02-20:
And these are the rules in `60-heimdall.rules`:
```
SUBSYSTEM=="usb", ATTR{idVendor}=="04e8", ATTR{idProduct}=="6601", MODE="0666"
SUBSYSTEM=="usb", ATTR{idVendor}=="04e8", ATTR{idProduct}=="685d", MODE="0666"
SUBSYSTEM=="usb", ATTR{idVendor}=="04e8", ATTR{idProduct}=="68c3", MODE="0666"
```
The `idProduct` value of the tablet is the same in the file.

Grimler91, 2022-02-20:
That all looks normal. Not really sure what issue is here, but doesn't really seem to be a heimdall issue. Libusb fails to open the device, maybe there is some issue with libusb, or maybe with some hardware (usb cable, usb port, ..)

If you have the possibility you could try from another distro, or from another computer and see if it works in that case

dfarmilant, 2022-02-22:
@Grimler91 

Hello again. I'm running Linux Mint 20.3 Una on a separate machine. I ran the logs and they have different outputs, though it still doesn't work. I'm going to read through them and refer to known issues to see if there's a solution. In the event I don't find anything, I'll tag you to determine next steps. Thanks for the help so far!

Grimler91, 2022-02-23:
@dfarmilant what's the error on the other machine?

If it at least succeeds to initialize the device, but fails with something else that might be a heimdall issue, then you could also try updated heimdall from here: https://git.sr.ht/~grimler/Heimdall/ and see if it works better

mkesper, 2022-12-16:
I seem to get the same behaviour with Debian bookworm (amd64, Debian build). The detected usb port seems to always be off by one. That's why it can't be opened with or without sudo. Will try the new version, too.

mkesper, 2022-12-16:
```
$ heimdall flash --BOOT 20221216-0235-postmarketOS-edge-phosh-22.1-samsung-m0-boot.img --verbose
Heimdall v1.4.2

Copyright (c) 2010-2017 Benjamin Dobell, Glass Echidna
http://www.glassechidna.com.au/

This software is provided free of charge. Copying and redistribution is
encouraged.

If you appreciate this software and you would like to support future
development please consider donating:
http://www.glassechidna.com.au/donate/

Initialising connection...
Detecting device...
libusb: error [get_usbfs_fd] libusb couldn't open USB device /dev/bus/usb/001/006, errno=5
ERROR: Failed to access device. libusb error: -1
~/Downloads [1]$ ls -hal /dev/bus/usb/001
insgesamt 0
drwxr-xr-x  2 root root       160 16. Dez 22:19 ./
drwxr-xr-x  4 root root        80 16. Dez 22:15 ../
crw-rw-r--  1 root root    189, 0 16. Dez 22:15 001
crw-rw-r--+ 1 root root    189, 1 16. Dez 22:15 002
crw-rw-r--  1 root root    189, 2 16. Dez 22:15 003
crw-rw-r--  1 root root    189, 3 16. Dez 22:15 004
crw-rw-r--  1 root root    189, 4 16. Dez 22:15 005
crw-rw----+ 1 root plugdev 189, 6 16. Dez 22:19 007

```

mkesper, 2022-12-16:
Replugged the USB cable:```

$ heimdall flash --BOOT 20221216-0235-postmarketOS-edge-phosh-22.1-samsung-m0-boot.img --verbose
Heimdall v1.4.2

Copyright (c) 2010-2017 Benjamin Dobell, Glass Echidna
http://www.glassechidna.com.au/

This software is provided free of charge. Copying and redistribution is
encouraged.

If you appreciate this software and you would like to support future
development please consider donating:
http://www.glassechidna.com.au/donate/

Initialising connection...
Detecting device...
libusb: error [get_usbfs_fd] libusb couldn't open USB device /dev/bus/usb/001/008, errno=5
ERROR: Failed to access device. libusb error: -1
~/Downloads [1]$ ls -hal /dev/bus/usb/001
insgesamt 0
drwxr-xr-x  2 root root       160 16. Dez 22:21 ./
drwxr-xr-x  4 root root        80 16. Dez 22:15 ../
crw-rw-r--  1 root root    189, 0 16. Dez 22:15 001
crw-rw-r--+ 1 root root    189, 1 16. Dez 22:15 002
crw-rw-r--  1 root root    189, 2 16. Dez 22:15 003
crw-rw-r--  1 root root    189, 3 16. Dez 22:15 004
crw-rw-r--  1 root root    189, 4 16. Dez 22:15 005
crw-rw----+ 1 root plugdev 189, 8 16. Dez 22:21 009
```

mkesper, 2022-12-16:
Issue persists even with Heimdall v2.0.2
```

$ uname -a
Linux debian 6.0.0-5-amd64 #1 SMP PREEMPT_DYNAMIC Debian 6.0.10-2 (2022-12-01) x86_64 GNU/Linux
Package: libusb-1.0-0
Version: 2:1.0.26-1
```
How can we further debug this?

ptesarik, 2023-01-04:
I believe I'm running into the same issue with openSUSE Tumbleweed. The device number is always off by one, because the device is reset and it gets a new ID on a reconnect. I mean, I can link this `heimdall` output:

```
[ 0.015339] [000059c9] libusb: debug [libusb_open] open 1.19
[ 0.448200] [000059c9] libusb: error [get_usbfs_fd] libusb couldn't open USB device /dev/bus/usb/001/019, errno=5
[ 0.448231] [000059c9] libusb: debug [libusb_open] open 1.19 returns -1
ERROR: Failed to access device. libusb error: -1
[ 0.448278] [000059c9] libusb: debug [libusb_exit]  
```

to these messages in the kernel log:

```
[23683.840804] usb 1-4: reset high-speed USB device number 19 using xhci_hcd
[23684.001212] usb 1-4: USB disconnect, device number 19
[23684.157612] usb 1-4: new high-speed USB device number 20 using xhci_hcd
[23684.308514] usb 1-4: New USB device found, idVendor=04e8, idProduct=685d, bcdDevice= 2.1b
[23684.308524] usb 1-4: New USB device strings: Mfr=1, Product=2, SerialNumber=0
[23684.308527] usb 1-4: Product: SAMSUNG USB DRIVER
[23684.308529] usb 1-4: Manufacturer: SAMSUNG
```

That means, the device ID was originally 1.19, but it got a new number after a USB reset. What I don't know is what causes this USB reset.

ptesarik, 2023-01-04:
I have ftraced the USB reset functions with full stack trace, and I don't really have a clue about the result:

```
        heimdall-25444   [002] ..... 25626.641981: <stack trace>
 => 0xffffffffc057b083
 => usb_reset_and_verify_device
 => usb_port_resume
 => usb_generic_driver_resume
 => usb_resume_both
 => __rpm_callback
 => rpm_callback
 => rpm_resume
 => __pm_runtime_resume
 => usb_autoresume_device
 => usbdev_open
 => chrdev_open
 => do_dentry_open
 => path_openat
 => do_filp_open
 => do_sys_openat2
 => __x64_sys_openat
 => do_syscall_64
 => entry_SYSCALL_64_after_hwframe
```

This pretty much looks like the kernel is resetting the device when the usbdev file is opened, but such behaviour should break pretty much all of libusb users…

FTR I'm running Linux 6.1.1 here.

ptesarik, 2023-01-04:
I had some luck after turning autosuspend off. Interestingly, I was not able to turn off autosuspend for this device specifically, but I had to turn it off globally:

`echo -1 > /sys/module/usbcore/parameters/autosuspend`

Since my device was already suspended at that point, I also had to wake it up once:

`echo -1 > /sys/bus/usb/devices/1-4/power/autosuspend`

I believe that the latter command should be sufficient, and it is a kernel bug if it isn't. I'll follow up with the kernel folks on that. It may in fact be the only issue here. I mean, if correct operation relied on disabling autosuspend for the target device, then it obviously won't work now.

ptesarik, 2023-01-04:
Back with news from the linux-usb mailing list. As we knew already, the implementation of the USB stack in download mode is crappy. In this case, the kernel sends a GET_STATUS control message to the device after resume to make sure the device is still there (as required by section 10.5.4.5 of the USB standard). The device should respond with two bytes of status in the data stage, but the Samsung download mode sends a zero-length data packet instead (i.e. no data).

Since this behaviour is hard-coded in the kernel resume logic, there is no way to resume from suspend without a disconnect and reconnect. That's why autosuspend cannot be disabled on that device when it is already suspended. OTOH udev might be fast enough to disable power control before the device is suspended.

In short, I propose to change the corresponding line in `60-heimdall.rules` to:

```
SUBSYSTEM=="usb", ATTR{idVendor}=="04e8", ATTR{idProduct}=="685d", MODE="0666", ATTR{power/control}="on"
```

mkesper, 2023-01-09:
Thanks a lot for investigating @ptesarik I will try that fix, too.

ptesarik, 2023-01-24:
Hey, @mkesper, did my proposed solution work for you?

mkesper, 2023-01-24:
Finally got to it. Yes, it works! NB: For Debian, the file is 
`/lib/udev/rules.d/60-heimdall-flash.rules`

Thanks a lot! :+1:

Kibubu, 2023-11-07:
Had the same issue on Manjaro with Heimdall 1.4.2

Added the proposed fix

`SUBSYSTEM=="usb", ATTR{idVendor}=="04e8", ATTR{idProduct}=="685d", MODE="0666", ATTR{power/control}="on"`

to 

`/lib/udev/rules.d/60-heimdall.rules`

Success