Benjamin-Dobell/Heimdall #510: v1.4.0 Not compatible with MacOS Monterey

NullVibes, 2022-03-20:
Package installer for MacOS (v1.4.0), from the glassechidna.com.au website, fails due to "incompatibility" with MacOS Monterey (v12.0.1).

[image]

jeromelefeuvre, 2022-03-31:
Same with 11.6.4 !

drpoutine, 2022-05-06:
Same error on M1 macs

0x77dev, 2022-06-26:
Any plans to support M1/Monterey in the future?

mys31f, 2022-08-24:
Seems like no devs are gonna reply to this thread... I don't even know either.

ne-vial, 2023-03-16:
Yeah, no possible to install on Ventura

mmoollllee, 2023-05-27:
I found [this bitbucket](https://bitbucket.org/benjamin_dobell/heimdall/downloads/?tab=downloads) downloaded `Heimdall-1.4.1-Unofficial-Signed.dmg` and was able to run heimdall cli on macos 13.3.1