Benjamin-Dobell/Heimdall #511: ERROR: Protocol initialisation failed on samsung SM-700f

yigitako, 2022-04-07:
While I was trying to install lineage OS to my SM-J700F  I am getting protocol error I don`t know why this is caused every other command works wit my Heimdall 

 **C:\Users\yigit\Documents\heídmal>heimdall detect
Device detected**

princesstwi, 2022-05-14:
If you are still having problems, i see C: which means you are most likely using windows. I recommend just using Odin. You may also need usb drivers from samsung

jinetty, 2022-05-27:
I'm facing this issue too and i'm using Artix Linux.

starbrights, 2022-07-25:
> If you are still having problems, i see C: which means you are most likely using windows. I recommend just using Odin. You may also need usb drivers from samsung

Odin wants to flash a tar.md5 file, while heimdall uses an img (at least for recovery). How to convert an IMG to be work with ODIN? I found some tools but at least I get a forced close when flashing AP. (But the not working protocol - more exact bulk transfer) might be the reason for that crash too.