Benjamin-Dobell/Heimdall #512: Use winusb

mrb4a1370, 2022-04-14:
Hi guys.how to Use winusb instead of libusb?

amo13, 2022-04-15:
I think you can use [zadig](https://github.com/pbatard/libwdi) to change the driver. Download it from the releases page of the linked repo.
Connect your device in download mode, launch zadig and replace the driver.

mrb4a1370, 2022-04-18:
> I think you can use [zadig](https://github.com/pbatard/libwdi) to change the driver. Download it from the releases page of the linked repo. Connect your device in download mode, launch zadig and replace the driver.

No. I meant to use winusb in the project

helix-osdev, 2024-03-21:
> > I think you can use [zadig](https://github.com/pbatard/libwdi) to change the driver. Download it from the releases page of the linked repo. Connect your device in download mode, launch zadig and replace the driver.
> 
> No. I meant to use winusb in the project

It's easy! You read the docs and write the code. Super simple stuff!