Benjamin-Dobell/Heimdall #513: flash to AP (in odin) means what in heimdall

viisauksena, 2022-05-11:
hi,
many howtos in the internet refer to flash AP CSC .. in odin,
is there a handy table to get corresponding heimdall actions for the command line?
like flashing AP means flash --RECOVERY recover.img (if this is even right?!)
is it possible to add these hints in the linux readme?

Grimler91, 2022-05-12:
Typically you only need to use the AP slot to flash custom roms and similar, but for keeping other low level stuff up to date you might want to flash the latest BL, CSC and CP partitions when those are updated in the stock rom. 
Exactly what partitions that are part of an AP, BL, CP or CSC archive depends on the device. You can extract the AP, BL, CP and CSC .tar.md5 archives found in Samsung's stock rom to see what partitions each of them contains (just rename `.tar.md5` -> `.tar` and then extract with `tar xf`).

Samsung has for whatever reason grouped certain partitions in AP, CP, BL and CSC archives, probably to make it easier to organise updates between different phone variants for different regions. They could then create a new firmware with just an updated bootloader part by just swapping the BL archive in the previous zip file.

viisauksena, 2022-05-12:
so in my case (samsung A505fn) the proposed flash to AP .. 
would be flashing the RECOVERY and VBMETA from inside the proposed file just normal with heimdall, thanks a lot.
(an similar with other partition"groups" - just extract them and flash them accordingly)