Benjamin-Dobell/Heimdall #516: [SM-P610] Tab S6 Lite (Wi-Fi), error "Failed to access device. libusb error: -12"

that-fela, 2022-11-03:
I have been trying to install LineageOS on my S6 Tab Lite (Wi-Fi). However, Heimdall gives me the error:  
`ERROR: Failed to access device. libusb error: -12`
I've tried Windows, Ubuntu and Mint and they all give the same error. I have tried the following:
- Different USB ports
- Different USB cables
- Dongles and direct
- USB 2.0 and 3.0

using `heimdall print-pit --usb-log-level debug` returns:

```
[timestamp] [threadID] facility level [function call] <message>
--------------------------------------------------------------------------------
[ 0.006994] [00003198] libusbx: debug [windows_clock_gettime_threaded] hires timer available (Frequency: 10000000 Hz)
[ 0.006994] [000026e0] libusbx: debug [libusb_get_device_list] 
[ 0.007995] [000026e0] libusbx: debug [windows_get_device_list] allocating new device for session [3EE]
[ 0.008995] [000026e0] libusbx: debug [get_api_type] driver(s): USBHUB3
[ 0.008995] [000026e0] libusbx: debug [get_api_type] matched driver name against HUB API API
[ 0.008995] [000026e0] libusbx: debug [windows_get_device_list] allocating new device for session [1A2]
[ 0.008995] [000026e0] libusbx: debug [get_api_type] driver(s): USBHUB3
[ 0.008995] [000026e0] libusbx: debug [get_api_type] matched driver name against HUB API API
[ 0.008995] [000026e0] libusbx: debug [windows_get_device_list] allocating new device for session [1E1]
[ 0.008995] [000026e0] libusbx: debug [get_api_type] driver(s): USBHUB3
[ 0.008995] [000026e0] libusbx: debug [get_api_type] matched driver name against HUB API API
[ 0.008995] [000026e0] libusbx: debug [windows_get_device_list] allocating new device for session [3F8]
[ 0.009994] [000026e0] libusbx: debug [windows_get_device_list] allocating new device for session [BC]
[ 0.009994] [000026e0] libusbx: debug [init_device] got bus number from ancestor #2
[ 0.009994] [000026e0] libusbx: debug [init_device] found 1 configurations (active conf: 1)
[ 0.009994] [000026e0] libusbx: debug [cache_config_descriptors] cached config descriptor 0 (bConfigurationValue=1, 34 bytes)
[ 0.009994] [000026e0] libusbx: debug [init_device] (bus: 1, addr: 3, depth: 1, port: 7): '\\.\USB#VID_048D&PID_5702#5&201B4142&0&7'
[ 0.009994] [000026e0] libusbx: debug [windows_get_device_list] allocating new device for session [4D]
[ 0.009994] [000026e0] libusbx: debug [init_device] got bus number from ancestor #2
[ 0.009994] [000026e0] libusbx: debug [init_device] found 1 configurations (active conf: 1)
[ 0.009994] [000026e0] libusbx: debug [cache_config_descriptors] cached config descriptor 0 (bConfigurationValue=1, 59 bytes)
[ 0.009994] [000026e0] libusbx: debug [init_device] (bus: 1, addr: 6, depth: 1, port: 3): '\\.\USB#VID_046D&PID_C07E#49932F693335'
[ 0.009994] [000026e0] libusbx: debug [windows_get_device_list] found existing device for session [1E1] (0.0)
[ 0.009994] [000026e0] libusbx: debug [init_device] found 1 configurations (active conf: 1)
[ 0.009994] [000026e0] libusbx: debug [cache_config_descriptors] cached config descriptor 0 (bConfigurationValue=1, 25 bytes)
[ 0.009994] [000026e0] libusbx: debug [init_device] (bus: 1, addr: 4, depth: 1, port: 9): '\\.\USB#VID_05E3&PID_0608#5&201B4142&0&9'
[ 0.009994] [000026e0] libusbx: debug [windows_get_device_list] allocating new device for session [DA]
[ 0.009994] [000026e0] libusbx: debug [init_device] found 1 configurations (active conf: 1)
[ 0.009994] [000026e0] libusbx: debug [cache_config_descriptors] cached config descriptor 0 (bConfigurationValue=1, 67 bytes)
[ 0.009994] [000026e0] libusbx: debug [init_device] (bus: 1, addr: 43, depth: 1, port: 2): '\\.\USB#VID_04E8&PID_685D#5&201B4142&0&2'
[ 0.009994] [000026e0] libusbx: debug [windows_get_device_list] allocating new device for session [8]
[ 0.009994] [000026e0] libusbx: debug [init_device] found 1 configurations (active conf: 1)
[ 0.009994] [000026e0] libusbx: debug [cache_config_descriptors] cached config descriptor 0 (bConfigurationValue=1, 241 bytes)
[ 0.009994] [000026e0] libusbx: debug [init_device] (bus: 1, addr: 16, depth: 1, port: 12): '\\.\USB#VID_046D&PID_0AAA#000000000000'
[ 0.010995] [000026e0] libusbx: debug [windows_get_device_list] allocating new device for session [9C]
[ 0.010995] [000026e0] libusbx: debug [init_device] found 1 configurations (active conf: 1)
[ 0.010995] [000026e0] libusbx: debug [cache_config_descriptors] cached config descriptor 0 (bConfigurationValue=1, 59 bytes)
[ 0.010995] [000026e0] libusbx: debug [init_device] (bus: 1, addr: 7, depth: 1, port: 4): '\\.\USB#VID_045E&PID_07F8#6&28DBA815&0&4'
[ 0.010995] [000026e0] libusbx: debug [windows_get_device_list] found existing device for session [3F8] (1.0)
[ 0.010995] [000026e0] libusbx: debug [init_device] found 1 configurations (active conf: 1)
[ 0.010995] [000026e0] libusbx: debug [cache_config_descriptors] cached config descriptor 0 (bConfigurationValue=1, 25 bytes)
[ 0.010995] [000026e0] libusbx: debug [init_device] (bus: 1, addr: 2, depth: 1, port: 3): '\\.\USB#VID_05E3&PID_0605#5&201B4142&0&3'
[ 0.010995] [000026e0] libusbx: debug [windows_get_device_list] found existing device for session [1A2] (1.0)
[ 0.010995] [000026e0] libusbx: debug [init_device] (bus: 1, addr: 1, depth: 0, port: 0): '\\.\USB#ROOT_HUB30#4&17DA066&0&0'
[ 0.010995] [000026e0] libusbx: debug [windows_get_device_list] allocating new device for session [294]
[ 0.010995] [000026e0] libusbx: debug [init_device] found 1 configurations (active conf: 1)
[ 0.010995] [000026e0] libusbx: debug [cache_config_descriptors] cached config descriptor 0 (bConfigurationValue=1, 177 bytes)
[ 0.010995] [000026e0] libusbx: debug [init_device] (bus: 1, addr: 5, depth: 1, port: 10): '\\.\USB#VID_8087&PID_0AA7#5&201B4142&0&10'
[ 0.010995] [000026e0] libusbx: debug [discovered_devs_append] need to increase capacity
[ 0.011994] [000026e0] libusbx: debug [get_api_type] driver(s): HidUsb
[ 0.011994] [000026e0] libusbx: debug [get_api_type] matched driver name against HID API API
[ 0.011994] [000026e0] libusbx: debug [windows_get_device_list] found existing device for session [BC] (1.3)
[ 0.011994] [000026e0] libusbx: debug [get_api_type] driver(s): BTHUSB
[ 0.011994] [000026e0] libusbx: debug [get_api_type] lower filter driver(s): ibtusb
[ 0.011994] [000026e0] libusbx: debug [windows_get_device_list] found existing device for session [294] (1.5)
[ 0.011994] [000026e0] libusbx: debug [get_api_type] driver(s): usbccgp
[ 0.011994] [000026e0] libusbx: debug [get_api_type] matched driver name against Composite API API
[... 35 lines cut ...]
[ 0.012995] [000026e0] libusbx: debug [windows_get_device_list] setting composite interface for [9C]:
[ 0.012995] [000026e0] libusbx: debug [set_composite_interface] interface[1] already set - ignoring HID collection: \\.\HID#VID_045E&PID_07F8&MI_01&COL01#8&5A20878&0&0000
[ 0.012995] [000026e0] libusbx: debug [windows_get_device_list] setting composite interface for [4D]:
[ 0.012995] [000026e0] libusbx: debug [set_composite_interface] interface[1] already set - ignoring HID collection: \\.\HID#VID_046D&PID_C07E&MI_01&COL02#8&B2626C7&0&0001
[ 0.012995] [000026e0] libusbx: debug [libusb_unref_device] destroy device 1.0
[ 0.012995] [000026e0] libusbx: debug [libusb_get_device_descriptor] 
[ 0.012995] [000026e0] libusbx: debug [libusb_get_device_descriptor] 
[ 0.012995] [000026e0] libusbx: debug [libusb_get_device_descriptor] 
[ 0.012995] [000026e0] libusbx: debug [libusb_get_device_descriptor] 
[ 0.012995] [000026e0] libusbx: debug [libusb_unref_device] destroy device 1.3
[ 0.012995] [000026e0] libusbx: debug [libusb_unref_device] destroy device 1.6
[ 0.012995] [000026e0] libusbx: debug [libusb_unref_device] destroy device 1.4
[ 0.012995] [000026e0] libusbx: debug [libusb_unref_device] destroy device 1.16
[ 0.012995] [000026e0] libusbx: debug [libusb_unref_device] destroy device 1.7
[ 0.012995] [000026e0] libusbx: debug [libusb_unref_device] destroy device 1.2
[ 0.012995] [000026e0] libusbx: debug [libusb_unref_device] destroy device 1.1
[ 0.012995] [000026e0] libusbx: debug [libusb_unref_device] destroy device 1.5
[ 0.012995] [000026e0] libusbx: debug [libusb_open] open 1.43
[ 0.012995] [000026e0] libusbx: debug [unsupported_open] unsupported API call for 'open' (unrecognized device driver)
[ 0.012995] [000026e0] libusbx: debug [libusb_open] could not open device: LIBUSB_ERROR_NOT_SUPPORTED
ERROR: Failed to access device. libusb error: -12
[ 0.012995] [000026e0] libusbx: debug [libusb_unref_device] destroy device 1.43
[ 0.012995] [000026e0] libusbx: debug [libusb_exit] 
[ 0.012995] [000026e0] libusbx: debug [libusb_exit] destroying default context
```

that-fela, 2022-11-03:
Was resolved my using Odin. Renaming the the .img file to `recovery.img`. Then using 7zip, adding it to a .tar file.

Grimler91, 2022-11-03:
From where did you install heimdall? What version are you using?

I'm guessing logs above were captured in windows. What usb drivers did you use? Logs references libusbx, which is a very old variant of libusb, so best guess is that usb drivers are too old