Benjamin-Dobell/Heimdall #517: heimdall print-pit does work with Samsung a52q

akamaus, 2022-12-25:
I was following instructions on https://wiki.lineageos.org/devices/a52q/install, but decided to try heimdall instead of odin as I dont have access to a Windows machine.

I activated download mode by pressing up+down buttons and plugging the usb cord in. After that I got the following. I didn't try the actual flashing.

```
heimdall print-pit --verbose --no-reboot
Heimdall v1.4.2

Copyright (c) 2010-2017 Benjamin Dobell, Glass Echidna
http://www.glassechidna.com.au/

This software is provided free of charge. Copying and redistribution is
encouraged.

If you appreciate this software and you would like to support future
development please consider donating:
http://www.glassechidna.com.au/donate/

Initialising connection...
Detecting device...
      Manufacturer: "Samsung"
           Product: "SM6150"

            length: 18
      device class: 2
               S/N: 0
           VID:PID: 04E8:685D
         bcdDevice: 021B
   iMan:iProd:iSer: 1:2:0
          nb confs: 1

interface[0].altsetting[0]: num endpoints = 1
   Class.SubClass.Protocol: 02.02.01
       endpoint[0].address: 82
           max packet size: 0010
          polling interval: 09

interface[1].altsetting[0]: num endpoints = 2
   Class.SubClass.Protocol: 0A.00.00
       endpoint[0].address: 01
           max packet size: 0200
          polling interval: 00
       endpoint[1].address: 81
           max packet size: 0200
          polling interval: 00
Claiming interface...
Setting up interface...

Initialising protocol...
ERROR: libusb error -7 whilst sending bulk transfer. Retrying...
ERROR: libusb error -7 whilst sending bulk transfer. Retrying...
ERROR: libusb error -7 whilst sending bulk transfer. Retrying...
ERROR: libusb error -7 whilst sending bulk transfer. Retrying...
ERROR: libusb error -7 whilst sending bulk transfer. Retrying...
ERROR: libusb error -7 whilst sending bulk transfer.
ERROR: Failed to send handshake!ERROR: Failed to receive handshake response. Result: -7
ERROR: Protocol initialisation failed!

Releasing device interface...
```

amo13, 2022-12-25:
Did you run it with sudo? Where did you get the binary from?

Grimler91, 2022-12-25:
Possibly same issue as fixed by PR https://github.com/Benjamin-Dobell/Heimdall/pull/478

julianfoad, 2024-05-25:
Duplicate of #413? I just added some help there.