Benjamin-Dobell/Heimdall #528: Heimdall does not accept boot.img and boot.tar files as firmware packages.

Marietto2008, 2023-07-08:
Hello to everyone.

I'm trying to enable KVM on my SM-A600G (Samsung Galaxy A6 with exynos7870). This is the guide that I'm following :

https://github.com/raspiduino/a6lte-kvm

I've reached Step 1 of the section "installing",where he says to :

> Transfer the patched img file to your Windows PC (to use Odin, in Linux you need [Heimdall] https://github.com/Benjamin-Dobell/Heimdall)) and rename it to boot.img. Add the file to tar using 7-zip or tar command then load tar file to AP in Odin. Flash your phone and when it says PASS!, you are doing well!

He didn't explain which kind of tecnique and file is accepted by Heimdall,so I'm here to ask this. At the moment I have a boot.img file and a boot.tar files. These files aren't accepted by Heimdall. When I click on "Heimdall Firnware Package / Browse" they aren't available. So,what should I do to install the pathed image file to my phone using heimdall ? thanks.

Grimler91, 2023-07-08:
Flash the patched img file from command line with `heimdall flash --BOOT /path/to/patched-boot.img`, or (possibly, haven't tried) through GUI. No need to rename the file or put it into zip file with heimdall

Marietto2008, 2023-07-10:
(I'm running Ubuntu 22.04 installed within a WSL2 system on Windows 11).
I've installed Heimdall with the following commands :

# mv magisk_patched-26100_JpvSO.img boot.img
# apt install heimdall-flash-frontend
# heimdall flash --BOOT boot.img

Heimdall v1.4.2

Copyright (c) 2010-2017 Benjamin Dobell, Glass Echidna
http://www.glassechidna.com.au/

This software is provided free of charge. Copying and redistribution is
encouraged.

If you appreciate this software and you would like to support future
development please consider donating:
http://www.glassechidna.com.au/donate/

Initialising connection...
Detecting device...
ERROR: Failed to detect compatible download-mode device.

Grimler91, 2023-07-11:
Does windows automatically pass usb devices through to WSL? If not you'll have to set that up somehow.

Probably easier to use odin if you are on windows