Benjamin-Dobell/Heimdall #529: Samsung SM-A736B/DS - ERROR: Failed to send handshake!

sysadminpower2019, 2023-07-26:
Hello,

I am using heimdall v.1.42 and still getting this error.  I used the code from github to build it but it still doesn't work

bbourgeat, 2023-08-10:
Hello, 
Same problem for me. I'm trying to flash twrp on my device (Samsung S10e), I'm using heimdall and ubuntu 20.04 and I always get that :

```
heimdall flash --RECOVERY twrpS10.img --no-reboot
Heimdall v1.4.2

Copyright (c) 2010-2017 Benjamin Dobell, Glass Echidna
http://www.glassechidna.com.au/

This software is provided free of charge. Copying and redistribution is
encouraged.

If you appreciate this software and you would like to support future
development please consider donating:
http://www.glassechidna.com.au/donate/

Initialising connection...
Detecting device...
Claiming interface...
Setting up interface...

Initialising protocol...
ERROR: Failed to send handshake!ERROR: Protocol initialisation failed!

Releasing device interface...
```

I read [https://github.com/Benjamin-Dobell/Heimdall/issues/491](url) but looks like interesting pages have disappeared (error404). Please, someone could help ? thanks.

Geremia, 2024-02-18:
@red0bear How should udev be properly configured? 
In my case, `lsusb` shows:
```
Bus 005 Device 053: ID 04e8:685d Samsung Electronics Co., Ltd GT-I9100 Phone [Galaxy S II] (Download mode)
```
So the device in download mode is being detected.

julianfoad, 2024-05-25:
Duplicate of #413? I just added some help there.

red0bear, 2024-05-26:
@Geremia ... sorry i get a bit away from here .. is too much toxic and full scammers. I see there is a problem to configure and access on download . but libusb doenst or something is missing depending in OS. It fails both github and pre compiled stuff.

Geremia, 2024-05-27:
@red0bear  Sorry, I don't have the device anymore to help you test it. ☹