Benjamin-Dobell/Heimdall #532: CMake Error at heimdall-frontend/CMakeLists.txt:11 (find_package)

mariano-daniel, 2023-09-12:
`$ cmake --version cmake version 3.27.4`
```
$ brew --version
Homebrew 4.1.10-11-g502434c
Homebrew/homebrew-core (git revision 87b45ed1ec7; last commit 2023-09-10)
Homebrew/homebrew-cask (git revision 8acf42495f; last commit 2023-09-11)
```

When I run:

`cmake -G "Unix Makefiles" -DCMAKE_BUILD_TYPE=Release -DQt5Widgets_DIR=/usr/local/opt/qt5/lib/cmake/Qt5Widgets ..`

I get:

```
CMake Error at heimdall-frontend/CMakeLists.txt:11 (find_package):
  By not providing "FindQt5Widgets.cmake" in CMAKE_MODULE_PATH this project
  has asked CMake to find a package configuration file provided by
  "Qt5Widgets", but CMake did not find one.

  Could not find a package configuration file provided by "Qt5Widgets" with
  any of the following names:

    Qt5WidgetsConfig.cmake
    qt5widgets-config.cmake

  Add the installation prefix of "Qt5Widgets" to CMAKE_PREFIX_PATH or set
  "Qt5Widgets_DIR" to a directory containing one of the above files.  If
  "Qt5Widgets" provides a separate development package or SDK, be sure it has
  been installed.
```

However, I cannot find `Qt5WidgetsConfig.cmake` nor `qt5widgets-config.cmake` anywhere.

I googled to no avail but all the threads I found mention people finding these files and adding the dir to the env variable, but I cannot find these files.

Any help would be super appreciated"

Grimler91, 2023-09-18:
On many linux systems Qt5WidgetsConfig.cmake is provided by a package named qt5-qtbase or qt5-qtbase-dev, so check if you have one of those

mariano-daniel, 2023-09-20:
> On many linux systems Qt5WidgetsConfig.cmake is provided by a package named qt5-qtbase or qt5-qtbase-dev, so check if you have one of those

Hello & Thank you @Grimler91 ! I don't have any of those on MacOS neither I can find them on Brew.

depperson, 2023-11-08:
Brew installs Qt5 in a special location so you need to tell cmake where to find the libraries:
```
cmake -G "Unix Makefiles" -DCMAKE_BUILD_TYPE=Release -DQt5Widgets_DIR=/opt/homebrew/opt/qt@5/lib/cmake/Qt5Widgets ..
```
Also due to a newer version of cmake [(CMP0006)](https://cmake.org/cmake/help/v3.9/policy/CMP0006.html) you will need to add this line in heimdall-frontend/CMakeLists.txt and heimdall/CMakeLists.txt:
```
CMAKE_POLICY(SET CMP0006 OLD)
```

That makes cmake happy but then I run into problems during make:
```
In file included from /Library/Developer/CommandLineTools/SDKs/MacOSX14.0.sdk/usr/include/c++/v1/string:545:
/Library/Developer/CommandLineTools/SDKs/MacOSX14.0.sdk/usr/include/c++/v1/__functional/hash.h:184:12: error: no matching constructor for initialization of 'pair<unsigned long, unsigned long>'
    return pair<_Size, _Size>(__a + __z, __b + __c);
           ^                  ~~~~~~~~~~~~~~~~~~~~
```
This seems to be due to the C++ version? Maybe my gcc or Apple clang is too new?

mariano-daniel, 2023-11-08:
Thanks @depperson ! I will try this in the morning (02:10am here) and see if I get the same results!

depperson, 2023-11-08:
Check out https://github.com/fathonix/heimdall-osx-arm64 if all you need is an m1 compatible heimdall binary.

mariano-daniel, 2023-11-10:
Welp, I guess I won't be getting too far with Heimdall-suite

```
==> Downloading https://bitbucket.org/benjamin_dobell/heimdall/downloads/heimdal
==> Downloading from https://bbuseruploads.s3.amazonaws.com/benjamin_dobell/heim
######################################################################### 100.0%
==> Installing Cask heimdall-suite
==> Running installer for heimdall-suite with sudo; the password may be necessar
installer: Package name is Heimdall Suite 1.4.0
installer: Installing at base path /
installer: The install failed. (The Installer encountered an error that caused the installation to fail. Contact the software manufacturer for assistance. The package is attempting to install content to the system volume.)
==> Purging files for version 1.4.0 of Cask heimdall-suite
Error: Failure while executing; `/usr/bin/sudo -u root -E LOGNAME=trixie USER= trixie USERNAME= trixie -- /usr/sbin/installer -pkg /usr/local/Caskroom/heimdall-suite/1.4.0/Heimdall\ Suite\ 1.4.0.pkg -target /` exited with 1. Here's the output:
installer: Package name is Heimdall Suite 1.4.0
installer: Installing at base path /
installer: The install failed. (The Installer encountered an error that caused the installation to fail. Contact the software manufacturer for assistance. The package is attempting to install content to the system volume.)
```

I tried picking up the .pkg file from the Caskroom, allowing the install from Settings --> Security but it also failed with:

[image]

Not sure what else to do here. Compile from source? I really need this package in order to flash my mum's old Samsung Galaxy Tablet 😢