Benjamin-Dobell/Heimdall #534: Where do i download the firmware packages?

hoods-py, 2023-11-13:
Where do i download the firmware packages?

hoods-py, 2023-11-21:
Okay yes, that was my issue. I thought because of the wording that there were downloadable firmware packages somewhere.

abuyusif01, 2023-11-24:
welcome to the android world, where nothing make sense

TheClogger666, 2023-11-26:
> Where do i download the firmware packages?

You can usually find your devices firmware on sites like SamFW, etc. They have a huge selection of every device and carrier

su02u2, 2025-10-31:
https://raw.githubusercontent.com/Benjamin-Dobell/Heimdall/master/Linux/README

Scroll down the README and you will find.......

How to Create a Heimdall Firmware Package: 

    Firstly, Heimdall's firmware package format is just a regular TAR archive
    compressed with gzip. The only two real requirements are that a valid
    firmware.xml must be included (refer to Appendix A) and you can only
    include files (no directories, links etc.) As such if you'd like there is
    nothing preventing you from creating Heimdall packages manually. Of course
    Heimdall Frontend provides a simple user interface that takes care of all
    the hard work for you.