Benjamin-Dobell/Heimdall #537: Incompatible Version of Mac OS   ??

KHALED-MSOUTI, 2024-03-18:
My MACBOOK PRO 2019 with intel processor runs macOS Sonoma 14.3.1 when installing Heimdall it gives me this error 
"This package is incompatible with this version of macOS."
    "The package is trying to install content to the system volume. Contact the software manufacturer for assistance."
    help to fix please
in the picture there is the issue
[image]

helix-osdev, 2024-03-19:
If you learned to ACTUALLY read shit you'd see your OS version is newer than the tool supports.

acacioveit, 2024-04-17:
maybe you could use https://github.com/fathonix/heimdall-osx-arm64, they have a `x86_64`, `arm64` and a `universal` version

DinBan, 2025-04-23:
Thank you! For me it helps - latest arm64 version 2.0.2 works on my MacBook M1 with macOS 15.5. But universal version do not start at all.