Benjamin-Dobell/Heimdall #542: Heimdall does not use SET_CONFIGURATION, which breaks Samsung devices with Qualcomm SOCs

kauwua, 2024-05-23:
Hello,

When working on a Samsung Galaxy Z Flip3 (SM-F711U1) with @max-r-b, we could not flash the device using Heimdall compared to Odin. 

When the host sends "ODIN", the device should answer by "LOKI" but wasn't.

After some research using Wireshark and a USB sniffer, we noticed that the SET_CONFIGURATION request is missing and instead a SET_INTERFACE request is used.

We propose the following WIP patch, that solves the issue for Qualcomm-based devices but breaks the Mediatek ones apparently (like a SM-A225F (Samsung Galaxy A22)). `SetupDeviceInterface` should probably be called only for Mediatek devices.

The SET_CONFIGURATION request is mandatory because it puts the device in the "Configured" state before it is used (Universal Serial Bus Specification Revision 2.0, 9.1.1.5 Configured). While SET_INTERFACE also resets some of the internal state of the device, the device might only fully reset and prepare for requests when SET_CONFIGURATION is called.

Thanks

kauwua, 2024-05-23:
[image]

This is capture made with the USB sniffer, after adding a SET CONFIGURATION. The device still does not answer, and we had to remove SET INTERFACE as well for it to work.

Grimler91, 2024-06-28:
Nice investigation! 

Which libusb version did you test with, and on what OS? On my archlinux machine (with libusb 1.0.27) set_configuration seems to be run automatically when device is connected to the computer:
[image]

Looking at some of my older logs it seems to have been the same since at least beginning of 2021 (libusb 1.0.24). If set_configuration is not run automatically on all OS'es we should definitely add it though.

I guess you have a filter in wireshark screenshot so that responses are not shown? Are those `GET DESCRIPTOR`, `SET CONFIGURATION` and `SET INTERFACE` really done by heimdall? They look similar to what I get automatically when device is connnected (as in above screenshot). Could you share the .cap/.pcapng file(s) of failed and succeeded flashes? (In private if preferred, henrik at grimler dot se)

I mainly have exynos devices to test with, and two older qcom phones (msm8916-a5lte (2015) and galaxy s4 (2013)). Seems like flashing works fine with or without an added `libusb_set_configuration()`, with the patch https://git.sr.ht/~grimler/Heimdall/commit/2ecc98020c607631d1371c8596d20cce84586047.