Benjamin-Dobell/Heimdall #547: I am getting "Failed to confirm end of PIT file transfer!" error.

uncharted-user, 2025-03-09:
I entered the following command,
```
heimdall flash --repartition --resume --pit A52Q_EUR_OPEN.pit \
	 --PERSIST persist.img \
	 --MISC misc.bin \
	 --DSP dspso.bin \
	 --MODEM modem.bin \
	 --BOOT boot.img \
	 --RECOVERY recovery.img \
	 --DTBO dtbo.img \
	 --APNHLOS NON-HLOS.bin \
	 --VBMETA_SYSTEM vbmeta_system.img \
	 --SUPER super.img \
	 --PRISM prism.img \
	 --OPTICS optics.img \
	 --CACHE cache.img \
	 --OMR omr.img \
	 --USERDATA userdata.img \
	 --XBL xbl.elf \
	 --XBL_CONFIG xbl_config.elf \
	 --AOP aop.mbn \
	 --TZ tz.mbn \
	 --HYP hyp.mbn \
	 --ABL abl.elf \
	 --KEYMASTER km4.mbn \
	 --CMNLIB cmnlib.mbn \
	 --CMNLIB64 cmnlib64.mbn \
	 --DEVCFG devcfg.mbn \
	 --QUPFW qupv3fw.elf \
	 --TOOLSFV quest.fv \
	 --STORSEC storsec.mbn \
	 --VBMETA vbmeta.img \
	 --UEFISECAPP uefi_sec.mbn \
	 --IMAGEFV imagefv.elf \
	 --APDP apdp.mbn \
	 --SECDATA sec.elf \
	 --BKSECAPP bksecapp.mbn \
	 --VK vaultkeeper.mbn \
	 --HDM tz_hdm.mbn \
	 --TZ_KG tz_kg.mbn
```
The output was,
```
Heimdall v2.1.0

Copyright (c) 2010-2017 Benjamin Dobell, Glass Echidna https://glassechidna.com.au
Copyright (c) 2021-2024 Henrik Grimler
This software is provided free of charge. Copying and redistribution is encouraged.

Initialising connection...
Detecting device...
Claiming interface...
Beginning session...

Some devices may take up to 2 minutes to respond.
Please be patient!

Session begun.

Uploading PIT
ERROR: Failed to confirm end of PIT file transfer!
ERROR: PIT upload failed!

Ending session...
Rebooting device...
Releasing device interface...
```
I am using Grimler's fork.

uncharted-user, 2025-03-09:
Removing `--repartition` works, but I needed re-partitioning. However, my current and re-partition pit are the same. Am I doing anything wrong, or is it a bug?

LuciaSelvlysende, 2025-08-30:
> I received another report that re-partitioning is broken as well, so need to test it out and investigate.
> 
> Does it work with odin? The other reporter couldn't get it to work with odin either

I also ran into this issue while trying to flash stock firmware to a mostly unresponsive samsung-degaswifi (SM-T230NU). I tried it on odin4 and got this error:
```
lucia@arch-usb ~> sudo ./odin4 \
    -b BL_T230NUOVU0AOG1_CL5332174_QB5557805_REV03_user_low_ship.tar.md5 \
    -a AP_T230NUOVU0AOG1_CL5332174_QB5557805_REV03_user_low_ship.tar.md5 \
    -c CP_T230NUOVU0AOG1_CL5332174_QB5557805_REV03_user_low_ship.tar.md5 \
    -s CSC_BNN_T230NUBNN0AOG1_CL5332174_QB5557805_REV03_user_low_ship.tar.md5
[sudo] password for lucia: 
Check file : BL_T230NUOVU0AOG1_CL5332174_QB5557805_REV03_user_low_ship.tar.md5
Check file : AP_T230NUOVU0AOG1_CL5332174_QB5557805_REV03_user_low_ship.tar.md5
Check file : CP_T230NUOVU0AOG1_CL5332174_QB5557805_REV03_user_low_ship.tar.md5
Check file : CSC_BNN_T230NUBNN0AOG1_CL5332174_QB5557805_REV03_user_low_ship.tar.md5
Reboot into normal mode
/dev/bus/usb/001/007
/dev/bus/usb/001/007
Setup Connection
initializeConnection
Set Partition
Receive PIT Info
success getpit
Upload Binaries
timh.bin
Fail request receive -1
FAIL!
Fail uploadBinaries
```

porkmagus, 2026-11-11:
Root cause found, with a patch: #558.

**What's happening.** The PIT file is Samsung-signed, and the bootloader verifies that signature from the *end* of the transfer — its PIT handler calls the crypto core with `(buf, len - 0x100, buf + len - 0x100, …)`, i.e. the signature is the last 256 bytes of whatever it received.

For a Galaxy S7 (`HEROQLTE_USA_VZW.pit`) the file is **7676 bytes**:

```
16-byte header + 56 * 132-byte entries = 7408
+ 268-byte trailer (12 zero bytes + 256-byte RSA-2048 signature) = 7676
```

heimdall's `SendPitData()` doesn't send the file. It re-serialises it through `PitData::Pack()`, which writes the header and the entry table into a zeroed, 4096-byte-padded buffer — and stops there. The 268-byte trailer becomes zeros, so the device receives a structurally valid PIT with a dead signature, refuses to confirm the end of the transfer (exactly the `Failed to confirm end of PIT file transfer!` here), and the session aborts. Nothing is written, which is why it reads like a transport failure rather than a rejection.

This is not device-specific: it's the same `Pack()` path in 1.4.2 and in 2.1.0, which is why the report reproduces on a52q and on an SM-T230NU.

**Fix (#558).** Keep the bytes as read from disk and send them verbatim at their real length instead of re-serialising.

Verified on a Galaxy S7 SM-G930V (heroqlte, MSM8996, bootloader rev 11, locked): before, `--repartition --pit` fails at the PIT with nothing written; after, `PIT upload successful` and a 22-partition Samsung-signed image set then flashes in a single session on a freshly repartitioned table.

Worth porting to the 2.x tree at https://git.sr.ht/~grimler/Heimdall — the PIT path there is unchanged, so it should apply as-is. Also worth noting for anyone hitting this: because the failing case writes nothing, retrying costs only a re-entry into download mode.